sparkfun/SparkFun_BNO08x_Arduino_Library #2: MCU keeps resetting the sensor

tomy983, 2023-09-10:
### Subject of the issue
The MCU keeps resetting the sensor if enabling multiple reports

### Your workbench
* What development board or microcontroller are you using? esp32-s3-n16r8
* What version of hardware or breakout board are you using?  My own https://github.com/tomy983/BNO086-and-esp32-s3-brakeout
* How is the breakout board to your microcontroller?
* How is everything being powered? USB
* Are there any additional details that may help us help you?

### Steps to reproduce
Configure for SPI, enable multiple reports.

### Expected behavior
Reports enables

### Actual behavior
MCU keep resetting BNO086

Removing the resetCheck from enableReport() [here](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/blame/f47aa3e04ae805ab6f5dc6bdbf7c19c854b17b8b/src/SparkFun_BNO08x_Arduino_Library.cpp#L1172) fixes the issue.
Increasing the counter in [this line](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/blame/f47aa3e04ae805ab6f5dc6bdbf7c19c854b17b8b/src/SparkFun_BNO08x_Arduino_Library.cpp#L1514) unfortunately it is not enough.

lewispg228, 2023-09-12:
Hi @tomy983,
Thank you for bringing this to our attention. 

I was able to replicate this issue with a ESP32-WROOM-32E (The SparkFun IoT Redboard). I found a solution by adding in a 100ms delay at the end of the enableReports() function within the sketch. Like so:

```

void setReports(void) {
  Serial.println("Setting desired reports");
  if (myIMU.enableRotationVector() == true) {
    Serial.println(F("Rotation vector enabled"));
    Serial.println(F("Output in form i, j, k, real, accuracy"));
  } else {
    Serial.println("Could not enable rotation vector");
  }
  delay(100); // This delay allows enough time for the BNO086 to accept the new configuration and clear its reset status
}

```

I believe this slight delay is allowing the BNO086 to accept the new configuration and clear its reset status. FWW, I tried bringing this delay down a few times, and it seems that when it gets at or below 40ms, then the repetitive resetting issue comes back.

With only this addition of the delay of 100ms, I was able to leave the rest of the library alone. 

And so (at least with my setup using the ESP32-WROOM-32E) I'm not sure that the hal_wait_for_int() inside enableReports [at line 1172](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/blob/main/src/SparkFun_BNO08x_Arduino_Library.cpp#L1172) is causing the issue. 

I'm curious to know how this was solving the issue for you. Either way, would you be willing to try adding in this 100ms delay (and keeping the `hal_wait_for_int()` ) and see how it goes for you?

Let us know and thanks,
Pete

tomy983, 2023-09-13:
Hi Pete,
Adding the delay as you said fixed the issue. 
It looks like that no matter what I do, the first call to `getSensorEvent()` in the loop (in the setup does not) causes the sensor to reset. Which is actually desirable, since also the first call to enable the accelerometer report during setup fails. 
I've tried different ways, adding delay, waiting for interrupt, commanding wake, calling `setReports()` only in setup and only after  `getSensorEvent()`, but for me apparently the way is only let the first `setReports() `partially fail in setup and recall it (needed anyway because the reset) in the `loop()`. I hope I made myself clear..

**I am playing with this sensor this days, so if you need something tested or else, let me know.**
My ultimate goal is to have a reliable sensor reading and publish using micro-ROS at 100Hz an [imu_msg](http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Imu.html) and a mag_msg...

On the other hand... If I could ask... (maybe you have a better place where to ask), I have two boards practically identical, esp32-s3 and BNO086 on board and not much else, and on both I have the following issues:

- i2c operation at anything other than 100kHz is not reliable (possibly because I am using polling instead of interrupts with i2c as mandated in the datasheet for the BNO086? - **btw, I really hope this library will go towards that...**)
- SPI on both boards works fine, is fast and reliable. BUT after a minute or two, maybe after performing the calibration movements (facing cube, rotating..) but without any special commands (the code is exactly as the one below) and in a relatively clean enviroment (no magnets etc) **the GYRO swap rotation axis!!** I mean.. if I rotate the sensor along the z, after a while (maybe after a second at rest..) it shows as rotating along the X.  Of course the rotation vector also reflect this... I really don't know why or what causes it, but it happens only while using SPI. Could you give it a try? Maybe it's just me, but on two different boards.... hmmm...
Thank you

```cpp
#include <Arduino.h>
#include <SparkFun_BNO08x_Arduino_Library.h>

#define SPI
// #define I2C
#define I2C_FREQ 100000

BNO08x myIMU;

#ifdef SPI
SPIClass *vspi = NULL;
#define VSPI FSPI
#endif

#define VSPI_MISO_SDA 1
#define VSPI_SCLK_SCL 13
#define VSPI_MOSI_I2C_ADD 14
#define BNO08X_CS 11
#define BNO08X_INT 2
#define BNO08X_RESET 12
#define BNO08X_PS0 42
#define BNO08X_PS1 41

// internal copies of the IMU data
float ax, ay, az, gx, gy, gz, qx, qy, qz, qw, mx, my, mz;
byte linAccuracy = 0;
byte gyroAccuracy = 0;
byte magAccuracy = 0;
float quatRadianAccuracy = 0;
byte quatAccuracy = 0;

int bno_reset_count = 0;

unsigned long previousMillis = 0;
const long printInterval = 500;

void error_loop()
{
    while (1)
        ;
}

void printResetReasonName()
{
    const char *reasons[] = {
        "Not Applicable",
        "Power On Reset",
        "Internal System Reset",
        "Watchdog Timeout",
        "External Reset",
        "Other"};
    Serial.println("**********BNO08x reset**********");
    Serial.print("Reset n.");
    Serial.print(bno_reset_count);
    Serial.print(" with reason: ");
[... 148 lines cut ...]
            quatAccuracy = myIMU.getQuatAccuracy();
            quatRadianAccuracy = myIMU.getQuatRadianAccuracy();
            break;
        default:
            Serial.print("************************** Other Event: ");
            Serial.println(eventId);
            break;
        }
    }
    unsigned long currentMillis = millis();

    if (currentMillis - previousMillis >= printInterval)
    {
        previousMillis = currentMillis;
        char buf[200];
        sprintf(buf, "ax: %f, ay: %f, az: %f \ngx: %f, gy: %f, gz: %f \nqx: %f, qy: %f, qz: %f, qw: %f \nmx: %f, my: %f, mz: %f", ax, ay, az, gx, gy, gz, qx, qy, qz, qw, mx, my, mz);
        Serial.println(buf);
    }
    if (myIMU.wasReset())
    {
        printResetReasonName();
        setReports();
    }
}
```

lewispg228, 2023-11-10:
Hi @tomy983 ,
Glad to hear the fix worked for you. We just verified this on all three SPI examples, and pushed a new release with this delay included at the end of setReports.

Regarding the other two issues you mentioned in your last comment. We will dig into those asap. We should be able to test this out early next week with our hardware (ESP32 IoT Redboard and the breakout). 

Note, for best reliability, we recommend wiring up reset and INT. Our initial guess is that this could help with reliability at higher I2C clock speeds.

If you are still experiencing issues with these two other issues, would you mind filing them as separate issues on this github repo?

Thanks and we will report back soon with our testing results early next week.

lewispg228, 2023-12-01:
Hi @tomy983 ,
As the initial issue described in this issue has been resolved with the additional delay, we are going to close out this issue.

In terms of the I2S Speed. If any one comes across this conversation and is running into the same thing, we would recommend trying 350Khz, as that seems to work reliably with our setup (ESP32 IoT Redboard and QWIIC/INT/RST). Also note, there has been an issue using `setClock()`, and so the work around currently is to include it in the begin function like so: `Wire.begin(SDA, SCL, 350000);`

And we have not been able to replicate the Gyro axis swapping issue on our setup, so please let us know if you are still experiencing that and/or have found a solution.

Thanks,
Pete

closed 2023-12-01 by lewispg228