sparkfun/SparkFun_BNO08x_Arduino_Library #3: Library: Can't initialize two BNO086 on one I2C bus

iLoveAndyBaker, 2023-09-19:
Devices ACK on addresses 0x4A and 0X46, but library won't allow both to init. 

RP2040 (As KB2040)

### Steps to reproduce
Daisychain two BNO086 via Qwiic connectors, jump address pin on one unit, cut pullups on end unit.

Run example code: 
```#include <Wire.h>

#include "SparkFun_BNO08x_Arduino_Library.h"  // CTRL+Click here to get the library: http://librarymanager/All#SparkFun_BNO08x
BNO08x myIMU;
BNO08x yourIMU;

void setup() {
  Serial.begin(115200);
  delay(5000);
  Serial.println();
  Serial.println("BNO08x Read Example");

  Wire.begin();
  Wire.setClock(100000); //Increase I2C data rate to 400kHz

  if (myIMU.begin(0x4B,Wire,-1,-1) == false) {
    Serial.println("First BNO08x not detected at default I2C address. Check your jumpers and the hookup guide. Freezing...");
    //while (1)
      ;
  } else{
    Serial.println("First IMU found!");
  }

  if (yourIMU.begin(0x4A,Wire,-1,-1) == false) {
    Serial.println("Second BNO08x not detected at default I2C address. Check your jumpers and the hookup guide. Freezing...");
    while (1)
      ;
  } else{
    Serial.println("Second IMU found!");
  }

```

### Expected behavior
Both chips ACK and program continues with  begin == false. BUT...

### Actual behavior
Second to be init'd will ack, but begin returns false

```BNO08x Read Example
Device acknowledged at 75
First IMU found!
Device acknowledged at 74
Second BNO08x not detected at default I2C address. Check your jumpers and the hookup guide. Freezing...
```

I changed the library Sparkfun_BNO8x_Arduino_Library.cpp to call out the address that ACK'd 

```  	if (isConnected() == false) // Check for sensor by verifying ACK response
    	return (false); 

// CHANGED THIS SERIAL MESSAGE
    Serial.print(F("Device acknowledged at "));
    Serial.println(_deviceAddress);
    //delay(1000);

    _HAL.open = i2chal_open;
```

Re-ordering the initialization of the boards doesn't help. The one that goes SECOND fails. 

```
BNO08x Read Example
Device acknowledged at 74
First IMU found!
Device acknowledged at 75
Second BNO08x not detected at default I2C address. Check your jumpers and the hookup guide. Freezing...
```

iLoveAndyBaker, 2023-09-20:
Tried on separate I2C busses, first device to .begin  will initialize, second will not. Reorder and it's still the first one in the code will work, the next will not.

LazaroFilm, 2023-09-27:
My solution is to allow the code to try multiple times. I noticed that if may fail the first few times. 

``` C
  while (myIMU.begin() == false)
  {
    Serial.println("BNO08x not detected");
    delay(10);
  }
  Serial.println("BNO08x found!");
  setReports();
```
sometimes it never connects, I may look into setting an attempt counter that sends a full reset signal if it fails too many times. A manual reboot usually solves it.

lewispg228, 2023-11-15:
Hi @iLoveAndyBaker ,
Sorry, but this can only support a single device on the bus. See [here](https://github.com/ceva-dsp/sh2/blob/main/shtp.c#L30). This is due to the fact that the CEVA-DSP SH2 driver has some static methods and static variables. Not sure if this is even possible with the SHTP protocol, but if you wanted to, it might be worth [starting another issue on their repo](https://github.com/ceva-dsp/sh2/issues) requesting this feature.

closed 2023-11-15 by lewispg228

rah2501, 2023-11-28:
I raised an issue with the SH2 library: https://github.com/ceva-dsp/sh2/issues/9

Is the closure of this SparkFun_BNO08x_Arduino_Library issue a signal that you're unwilling to add support for more than one sensor?  Do you consider the limitation of running only a single sensor to be a bug that should be fixed?  If so, please reopen this issue.

lewispg228, 2023-11-29:
Hi @rah2501 , Thank you for submitting the issue and PR on the SH2 Library. When CEVA responds and/or adds in the ability to do multiple sensors, we would be excited to adapt our library to also work with multiple sensors. For now, we will leave this issue open. Thanks again and we look forward to seeing what CEVA says.

rah2501, 2024-10-09:
> we look forward to seeing what CEVA says

FYI, CEVA have completely ignored the github issue, the PR I raised and generally seem to have stopped communicating with the community.  I even did some digging and found the email address of one of the engineers (David Wheeler) and tried emailing them but again, no response.  Disappointing.

iLoveAndyBaker, 2024-10-10:
I have collected the following: In the SH2 library:
@rah2501 states that: The BNO08x library calls [sh2_service()](https://github.com/ceva-dsp/sh2/blob/b514b1e2586ddc195e553dac89fc94c637b25298/sh2.c#L1929) which uses [another global variable named _sh2](https://github.com/ceva-dsp/sh2/blob/b514b1e2586ddc195e553dac89fc94c637b25298/sh2.c#L429) for sensor-specific state.

@rah2501 did some changes to sh2.h at https://github.com/ceva-dsp/sh2/pull/10 but just copying that file over still wouldn't make it compile, and that's the extent of my expertise (I'm more of a project manager in this respect)

So even though it's probably not best practices to just patch up the SH2.h that lives in the Sparkfun library, I kinda don't see how this is going to get resolved in the usual proper manner without support from the upstream CEVA people....without simply patching up this version that's sitting here in the Sparkfun library.  There's nothing stopping me from doing this on my own hard drive either, but that's not very useful to anybody else. 

Oh and besides the whole static methods issue, there might need to be a need to call out the INT and RST per device too, because I don't see how #define BNO08X_INT  xxx is going to work unless it's common across this whole SHTP bus, which I confess I hardly understand its purpose.

iLoveAndyBaker, 2024-10-10:
CEVA says fork it and fend for ourselves. 

https://github.com/ceva-dsp/sh2/issues/9#issuecomment-2405479339

julkatpen, 2025-01-14:
Hello everyone, did you fork sh2 in order to fix this problem?

I am trying to connect 8 BNO086 via an I2C Multiplexer using the sparkfun library but i am running in to the same error that you described. I am a mechanical engineering student and writing my bachelor thesis, therefore i am sadly not able to write all of the code by myself. If anything worked out for you I would be very happy if you would share it with me.

rah2501, 2025-01-14:
> Hello everyone, did you fork sh2 in order to fix this problem?

No.

> If anything worked out for you I would be very happy if you would share it with me.

I shelved my project because of this issue.

Unless you have the software development resources to reimplement the driver, the BNO0xx chips are essentially useless for proper engineering.

julkatpen, 2025-01-16:
Thanks a lot for your response.
Do you think it should be possible to use the library if I only create one
sensor object and use only this one when I switch through the channels of
the multiplexer?
I tried doing that but it didnt work out, but I am not sure if it is
because of my lack of programming skill or because there would also be an
issue with the library. I am not sure if this would act as if I only had
one sensor connected.
If you have any idea on that I would be happy if you would let me know.
Thanks in advance

julkatpen, 2025-01-21:
I got it work for my purposes. I will post the fix so that anyone who has the same issue can maybe use it.

 The SparkFun_BNO08x_Arduino_Library library accesses the SH2 library, which uses the SHTP library, where there is a definition for MAX_INSTANCES (1) on line 148. When .begin() is applied to the sensor object, multiple steps in the SparkFun_BNO08x_Arduino_Library and the SH2 library result in the SHTP library requesting a new instance. However, since the old version only allows one instance, initialization fails, making usage impossible afterward. When this value is adjusted to the desired number of sensors to be operated via a multiplexer, initialization and data output work as expected.
It is possible that this modification may lead to issues in other operating modes and require further adjustments to the library.

Here is the exact process from MAX_INSTANCES() to .begin():

shtp.c:
   MAX_INSTANCES(1)
   used in getInstance
   used in shtp_open
sh2.c:
   used in sh2_open
SparkFun_BNO08x_Arduino_Library.c:
   used in _init
   used in begin
Your Code:
   uses .begin in initializeSensor

unusorin, 2025-02-05:
Amazing, placing blame on someone else. Just FYI this works fine on a raspberry pi and the python libraries.