sparkfun/SparkFun_BNO08x_Arduino_Library #4: How to use two IMUs with SPI?

joshgalvan, 2023-09-26:
Hello!

I'm unable to use two IMUs with SPI via myIMU1.begin() and myIMU2.begin(). I also haven't had any luck trying to use two different SPI ports on both of the IMUs. I'm using a Teensy 3.6, and can get the first IMU to work on the regular `SPI` bus, but am unable to initialize the same sensor on `SPI1`, despite having everything wired correctly.

Any suggestions?

My current setup is:
```
  if (myIMU2.beginSPI(BNO08X_2_CS, BNO08X_2_INT, BNO08X_2_RST, 900000, SPI1) == false) {
    Serial.println("myIMU2 not detected. Check your jumpers and the hookup guide. Freezing...");
    while (1);
  }
  Serial.println("myIMU2 found!");
  setReports2();

  if (myIMU.beginSPI(BNO08X_CS, BNO08X_INT, BNO08X_RST, 900000, SPI) == false) {
    Serial.println("myIMU not detected. Check your jumpers and the hookup guide. Freezing...");
    while (1);
  }
  Serial.println("myIMU found!");
  setReports();
```

joshgalvan, 2023-09-26:
I have found that by altering line 131 of `SparkFun_BNO08x_Arduino_Library.cpp` to `_spiSport = &SPI1;` the sensor is able to initialize and "be found", but then it is unable to enable a report, in my case `enableGameRotationVector(3)` fails.

lewispg228, 2023-11-15:
Hi @joshgalvan ,
Sorry, but this can only support a single device on the bus. See [here](https://github.com/ceva-dsp/sh2/blob/main/shtp.c#L30). This is due to the fact that the CEVA-DSP SH2 driver has some static methods and static variables. Not sure if this is even possible with the SHTP protocol, but if you wanted to, it might be worth [starting another issue on their repo](https://github.com/ceva-dsp/sh2/issues) requesting this feature.

closed 2023-11-15 by lewispg228