sparkfun/SparkFun_BNO08x_Arduino_Library #5: Tare X and Y axis only.

LazaroFilm, 2023-09-27:
### Subject of the issue
I would need to tare the X and Y axis without changing the Z axis, but the library only allows for Z or ALL axis to be zeroed out. Setting a Tare on all stops my device from working properly.
Looking at the SH2 code, it seems tat you can select the axis you want to tare with a binary/Hex code but it's not exposed in the Sparkfun library.

### Your workbench
RP2040 MCU with BNO085 module via I2C running under Arduino-pico

### Steps to reproduce
Set a Tare, only option is bool for Z or ALL. 

### Expected behavior
Select X, Y, Z, XY, XZ, YX, ALL tare options.

### Actual behavior
It works without tare, but if the sensor is not perfectly mounted an error will be forwarded to my code.

lewispg228, 2023-11-10:
Hi @LazaroFilm ,

I started working on this on a new branch called "[tare_individual_axis](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/tree/tare_individual_axis)", but haven't had any luck yet. 

Even with these modifications, the sensor only seems to react to the Z or ALL options. Datasheet for the BNO08x states on [page 42](https://docs.sparkfun.com/SparkFun_VR_IMU_Breakout_BNO086_QWIIC/assets/component_documentation/BNO080_085-Datasheet_v1.16.pdf) that there are only these two types of tare. However, the CEVA Driver "[sh2_setTareNow()](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/blob/main/src/sh2.c#L2071)" function comments might lead one to think that you can specify other axis combinations.

I will reach out to see if we can get more info about this issue.

LazaroFilm, 2023-11-10:
Hi @lewispg228 , 
I've been down the same rabbit hole it seems. The sh2 docs does mention each 3 axes separately addressable by individual bits, but it also only mentions XYZ and Z only as tare. My current workaround has been to take the Euler results for XYZ, then only sending X and Y to a euler conversion, keeping Z = 0. Then sending this modified quaternion struct coordinates to `myIMU.setReorientation(quaternion)` instead of the standard Tare.

lewispg228, 2023-11-10:
Hi @LazaroFilm ,
Thanks for sharing your solution. We hope this can serve as a work-around for you and any others that need this capability.

FWW, it looks like there was some work on the [CEVA-dsp\sh2 driver around the sh2_TareNow() function](https://github.com/ceva-dsp/sh2/commit/74f1d99d46c8528671e1d653379a1dea91610ce0), but I'm not sure this is directly related to this specific issue (the bits set on the additional masks don't seem make sense yet). 

I will attempt to update our library to use the latest release of the SH2 driver (v1.4.0) and see if this helps out in anyway. I'll report back here if I am able to get individual axis to tare properly.

LazaroFilm, 2023-11-10:
Thanks!
And thank you for bringing a fairly exhaustive library to interface with this powerful sensor. I'm really happy with the quality vs the BNO055 i had before and other sensors where the fusion had to be done in the MCU, taking most of the chip capacity.

lewispg228, 2023-11-13:
Hi @LazaroFilm ,
Just a quick update here, we started bringing in the latest ceva-dsp sh2 driver (v1.4.0). Most things are compiling/working, however, testing out all of the examples is still in progress. It is in this branch:

[https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/tree/ceva-dsp-sh2-v1.4.0](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/tree/ceva-dsp-sh2-v1.4.0)

Unfortunately, this update has not changed the behavior regarding the tareNow fuction. Sending the individual arguments to tare only x or y does not seem to cause anything to happen. 

We will continue to dig into this issue, but just wanted to give you an update. You are welcome to pull the branch with the updated sh2 driver and try it on your hardware, if you like.

Thanks for your patience as we iron these things out and let us know how it goes.

LazaroFilm, 2023-11-17:
So it can't be done but it looks like Game Mode may be an option for my issue. 

https://github.com/ceva-dsp/sh2/issues/8

closed 2023-11-17 by LazaroFilm