sparkfun/SparkFun_BNO08x_Arduino_Library #6: SparkFun Thing Plus - SAMD51 loses com port after programming with examples

mkinney88, 2023-10-04:
SparkFun Thing Plus - SAMD51 loses com port after programming with SparkFun VR IMU BNO08X examples.

I have the SparkFun Thing Plus SAMD51 (Qwiic) and the SparkFun VR IMU Breakout - BNO086 (Qwiic). I downloaded all the required libraries. When i program the SAMD51 after the program uploads successfully the board reboots out of bootloader and then windows gives me a notification "USB device not recognized" and does not display a com port. If I upload the Adafruit BNO08x example it works as expected.

mkinney88, 2023-10-04:
I was able to solve the issue if you move the #include <Wire.h> to after the #include "SparkFun_BNO08x_Arduino_Library.h". I am not 100% sure why this is but I would assume it has to do with something in the SparkFun_BNO08x_Arduino_Library.h. Hope this helps anyone that runs into this problem.

lewispg228, 2023-11-10:
Hi @mkinney88 ,
Thanks for making us aware of this issue. Glad you were able to find a work-around, but that does seem strange. We intend to try to duplicate this issue with hardware next week. 

Maybe the micro is trying to print to the serial terminal before it is done enumerating the com port. Another solution that might be worth trying would be to put a while loop in the setup to wait for Serial to return true. Something like this:

`while (! Serial); `

lewispg228, 2023-11-14:
Hi @mkinney88 ,
We were able to test out some hardware today with some success! We started a branch with the [necessary tweaks](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/commit/d62224f619aa730f7147e5e643283f81b11a10af) to the library to work with a SAMD51 Thing Plus.

Three things needed to change:

1. The arguments sent to the begin function needed to be signed ints. Previously, they were incorrectly uints, and therefore not performing their duty and causing the functions that use them to hang up. We are not entirely sure how this hasn't caused issues on other architectures yet, but it's probably something to do with how various boards packages implement the types of int8_t and uint8_t.
2. `while(!Serial);` was added to the top of setup(). Waiting for Serial to become alive is necessary for the native USB to enumerate properly.
3. Wiring up RST and INT control lines from the SAMD51 gpio pins to the sensor. Also, defining them at the top of the example and passing them to the begin function so the library instance can use them to have tighter control of the communication timing. Without these, the example continues to fail at myIMU.begin().

As a side note, we tried your suggestion to move the #include <wire.h> (without any of the above modifications), and although this did allow the com port to enumerate properly, the example would still not print anything to the terminal. We are still unsure as to how this slight change is actually effecting the timing/ability for the native USB to enumerate.

The following branch is working for example 1. We will continue to update this branch to get the other examples working. 

[https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/tree/samd51_support](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/tree/samd51_support)

Note, this example used with the SAMD51 does require wiring up INT and RST from GPIO pins to the sensor. For the example 1 code, we have chosen pins 4 and 22:

```
#define BNO08X_INT  4
#define BNO08X_RST  22
```

And these (along with the address and wire port) are passed into the begin function like so:

```
  if (myIMU.begin(BNO08X_ADDR, Wire, BNO08X_INT, BNO08X_RST) == false) {
    Serial.println("BNO08x not detected at default I2C address. Check your jumpers and the hookup guide. Freezing...");
    while (1)
      ;
  }
  Serial.println("BNO08x found!");
```

If you could try this on your setup that would be very helpful. Let us know and thanks!

We'll report back as we get the remaining examples updated.
Thanks,
Pete

lewispg228, 2023-11-15:
Testing out the examples is coming along. the following "checked" examples have been verified with a SAMD51.

- [x] 1
- [x] 2
- [x] 3
- [x] 4
- [x] 5
- [x] 6
- [x] 7
- [ ] ~~8 Advanced Config~~
- [x] 9
- [ ] ~~10 Timestamp~~
- [ ] 11 _Data looks incorrect and timing seems to be somewhat inconsistent - not sure what's going on here yet._
- [x] 12
- [x] 13
- [x] 14
- [x] 15
- [x] 16
- [x] 17
- [x] 18
- [x] SPI 1 _Note the CS pin needs to be changed to an available GPIO on the SAMD51 Thing+_
- [x] SPI 2 _Note the CS pin needs to be changed to an available GPIO on the SAMD51 Thing+_
- [x] SPI 3 _Note the CS pin needs to be changed to an available GPIO on the SAMD51 Thing+_

Example_08_AdvancedConfig is not currently working on SAMD51. To use alternate pins for I2C on the SAMD51, it gets a little more complex (involving SERCOM stuff). This example is still working on the ESP32 IoT Redboard, so we are going to leave this alone for now. 

If anyone comes across this comment and wants to pursue the AdvancedConfig example on a SAMD51, then we would recommend they check out our tutorial on utilizing SERCOM on SAMD Boards:

[https://learn.sparkfun.com/tutorials/adding-more-sercom-ports-for-samd-boards/all#adding-an-i2c](https://learn.sparkfun.com/tutorials/adding-more-sercom-ports-for-samd-boards/all#adding-an-i2c)

Also note, Example_10_TimeStamp has an issue with the SAMD51 Arduino Boards package not being capable of printing a uint64_t (the timestamp type variable).

Because most examples are working now, we will go ahead and push these changes to the branch and soon after merge into main. 

Hope this is helpful to you (and other SAMD51 users), please let us know how your setup is going,

Thanks,
Pete

lewispg228, 2023-11-17:
SAMD51 support added in [release v1.0.3](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/releases/tag/v1.0.3).

closed 2023-11-17 by lewispg228