sparkfun/SparkFun_BNO08x_Arduino_Library #7: Calibration removed?

LazaroFilm, 2023-10-04:
Why was the calibration removed from this library? 
I see the Tare, but one of my sensors is out of alignment between the accelerometer and the gyroscope and needs to be calibrated before a tare.

lewispg228, 2023-11-30:
Hi @LazaroFilm ,
When the BNO086 Breakout came out, we migrated this library to use the SH2 Driver. The calibration example was left out because we were unable to easily port it over and we needed to launch the product with a mostly functioning library.

We started working on re-creating the [Calibration example on a branch here](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/blob/Calibration_example/examples/Example_19_Calibrate/Example_19_Calibrate.ino). It is currently stalling out at the function sh2_startCal(), and we're not sure what's going on there. We will reach out to ceva-dsp and see if they can help with this. 

Of course, you would also be very welcome to take a look and let us know if you see anything.

Thanks and we'll keep ya posted,
Pete

lewispg228, 2023-11-30:
Just filed an issue on the sh2 driver about this. Hope to hear back from them soon. https://github.com/ceva-dsp/sh2/issues/11

lewispg228, 2023-12-01:
We were trying to use a function "sh2_startCal()" that is not supported by the BNO086. It looks like dynamic calibration is on by default for the Accel and Mag. And it stores the DCD to RAM every 5 seconds, so the "save" feature of the example may not really be necessary for most appilcations.

In light of this, we have changed the focus of the example to demo how to control the configurations. That is, now it shows how to turn dynamic calibration on/off for each individual sensor (accel/gyro/mag).

We also asked about the interactive calibration features of the BNO086, as we can't seem to find more info about this. That might lead to another example, however, we feel that most users will most likely be fine using dynamic calibration and following along with the positions/motions described in their [calibration procedure document here](https://docs.sparkfun.com/SparkFun_VR_IMU_Breakout_BNO086_QWIIC/assets/component_documentation/BNO080-BNO085-Sesnor-Calibration-Procedure.pdf).

Cheers,
Pete

closed 2023-12-01 by lewispg228

RndyP, 2024-03-11:
IMHO, the calibration procedure for the BNO devices is as clear as mud. See:
https://community.robotshop.com/forum/t/cmps14-how-does-calibration-work/104008