sparkfun/SparkFun_BNO08x_Arduino_Library #10: Enable GeoMagnetic Rotation Vector - Code reference provided

SFSailor, 2023-11-08:
### Subject of the issue
The Sparkfun library does not expose the geomagnetic rotation vector, needed to access the IMU's native compass calculation

### Your workbench
BNO086 + MEGA

### Steps to reproduce
Run the Euler example. Note the yaw. Rotate the unit. Note the non-zero yaw. Restart unit. Yaw is now 0, not mag heading.

### Expected behavior
Should be mag heading

### Actual behavior
Resets to zero

I have documented small changes needed to fix this in the Sparkfun forum

https://forum.sparkfun.com/viewtopic.php?f=83&t=60523&p=245145#p245145

lewispg228, 2023-11-17:
Hi @SFSailor ,
Thank you for bringing this to our attention. We have implemented your idea in a 19th I2C example on a new branch of the library located here:

[https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/tree/geomagnetic_rotation_vector](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/tree/geomagnetic_rotation_vector)

From our initial tests (using an ESP32 IoT Redboard), it seems to be performing properly and holding a consistent yaw reading through resets/power-cycles. 

Would you mind reviewing [this code branch](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/blob/geomagnetic_rotation_vector/examples/Example_19_GeomagneticRotationVector_to_EulerAngles/Example_19_GeomagneticRotationVector_to_EulerAngles.ino) and possibly verifying it on your setup?

We would also like to give you proper attribution for this work. Currently, we have put your github username in the [top header comment](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/blob/geomagnetic_rotation_vector/examples/Example_19_GeomagneticRotationVector_to_EulerAngles/Example_19_GeomagneticRotationVector_to_EulerAngles.ino#L7). Please let us know if that is sufficient.

Thanks again and we look forward to your reply,
-Pete

SFSailor, 2023-11-27:
Thanks!

I have been out of town and will give it a shot!

Michael

lewispg228, 2023-12-01:
Hi @SFSailor ,
Hope you are doing well. 

As we have verified this on our setup (ESP32 IoT Readboard using QWIIC/INT/RST), we have decided to move forward with this example. We just [pulled the branch into main](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/pull/16) and will be closing out this issue. 

If you do get a chance to verify this example on your setup, please let us know. 
Thanks and have a great weekend!
-Pete