sparkfun/SparkFun_BNO08x_Arduino_Library #18: Gyro misalignment issue in RotationVector and GameRotationVector

LazaroFilm, 2023-12-26:
I'm having some issues with gyro misalignment when getting the fused sensor values. 
If I rotate the sensor around the Z axis, the X and Y axes go off level during the rotation by about 1 degree up to 5 degrees. After stopping the rotation, the X and Y axes get corrected and come back to true values. It seems to me that the Gyroscope is misaligned with the accelerometers and try to rotate the quaternion with a misaligned offset. 
Anyone else with a similar issue?

mehmetdrnmrkt, 2024-01-09:
Hello, I have the same problem, did you find any solution?

lewispg228, 2024-01-09:
Hi @LazaroFilm and @mehmetdrnmrkt , Thanks for bringing this issue to our attention. We are currently looking into this. We'll keep you posted if we can replicate the issue on our end and find a solution. Just double-checking, but have you ensured a proper Tare on the Z axis happens prior to your testing? -Pete

LazaroFilm, 2024-01-09:
I'll investigate more regarding the tare. I have a few leads that came up from troubleshooting. The issue seems to be more pronounced when powering the device and the McU from a buck converter at 5v. When powering the MCU from USB and powering the BNO from the 5v line of the MCU, it works better. 

I also noticed that sometimes I'm able to get rid of the misalignment by rotating the IMU 180° then wait for it to correct itself over and over (maybe 10+ times) until the offset gets corrected by the algorithm.

lewispg228, 2024-01-09:
@LazaroFilm Thanks for sharing that new info. Please let us know how it goes with the Tare. 

Also, are you truly powering the BNO086 with 5V? It's max is 3.6V, so I'm not sure if that is playing into this. 

Another thought on power supplies, not sure what MCU/dev board you are using, but most of our dev boards have a linear regulator producing the 3.3V output, and this should be much less noisy than any buck regulator. We're not entirely sure at this point, but this could play into the differences you are seeing.

LazaroFilm, 2024-01-13:
I switched to 3.3v and it seems to have solved most of the issue. However I think some of my BNO08X IMU boards still have some misalignment, I believe feeding them over 5v may have permanently damaged those. I'm ordering new ones and will report if fresh sensors don't have the same issue.

LazaroFilm, 2024-01-18:
Now powered by 3.3v. new module and I still get the issue. Here is the visualization through X-IMU from quaternion coordinates.

[image]