sparkfun/SparkFun_BNO08x_Arduino_Library #19: what prevents SPI to function on a Micromod machine learning + Artemis MCU?

diplodocuslongus, 2024-05-06:
### BNO086 not detected on MicroMod + artemis using SPI

On a Machine Learning MicroMod board with an Artemis processor, my BNO086 SF breakout board is not detected.
I have verified and re-verified (and many more times) that connections and pin definitions are correct, but the BNO086 remains undetected.

The same HW connection and code (with appropriate adjustment for the pin definition) works right  away on a teensy41 board.

Is there anything specific to adjust (HW or Software) for that particular setup (artemis MCU + MicroMod Machine Learning)?

### My setup

* Artemis MCU (MicroMod processor board)
* MicroMod Machine Learning board
* Powered by USB

### Steps to reproduce

Hardware connection:

MicroMod Machine Learning --> BNO086
 CS  --> CS
  COPI --> SI
  CIPO --> SO
  SCK  --> SCK
  DO  --> INT
  D1  --> RST
  3V3  --> 3V3
  GND  --> GND

  BNO086 "mode" pins set for SPI:
  PSO --> 3V3
  PS1 --> 3V3

Code is the [Example_01_SPI_RotationVector.ino](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/blob/main/examples/SPI/Example_01_SPI_RotationVector/Example_01_SPI_RotationVector.ino) with the only modifications (to match hardware connection) being as follows:

```
#define BNO08X_CS SPI_CS // also tested with A0 and 55 
#define BNO08X_INT D0 // tested to work fine in I2C mode for the sketches requiring INT and RST
#define BNO08X_RST D1 // tested to work fine in I2C mode for the sketches requiring INT and RST

```

### Expected behavior

```
BNO08x Read Example
BNO08x found!
Setting desired reports
Could not enable rotation vector
Reading events
sensor was reset Setting desired reports
Rotation vector enabled
Output in form i, j, k, real, accuracy
sensor was reset Setting desired reports
Rotation vector enabled
Output in form i, j, k, real, accuracy
-0.11,0.02,0.65,0.75,3.14
-0.11,0.02,0.65,0.75,3.14

```

### Actual behavior

```
BNO08x Read Example
BNO08x not detected. Check your jumpers and the hookup guide. Freezing...

```

lewispg228, 2024-07-25:
Hi @diplodocuslongus , Sorry for such a delayed response here. Are you still having trouble getting the BNO086 to be detected over SPI? 

My first thought is that the SPI pins are somehow getting set incorrectly/differently between boards packages, but I could be wrong. I should be able to get some time to look further into this over the next few days. 

Let us know and again, sorry for the delay here. 
-Pete

lewispg228, 2024-07-29:
Hi @diplodocuslongus ,
SPI works with the following pins:

```
#define BNO08X_CS   PWM0
#define BNO08X_INT  D0
#define BNO08X_RST  D1
```

When selecting which pins you'd like to use for CS, INT, and RST, keep this in mind:

We can't use SPI_CS, because that pin is only routed to the CS pin for the uSD card. 

[image]

D0 is also used as the CS pin for the on-board camera SPI. In this way, it's not the most ideal pin to use for the CS on an external BNO086. Note, D0 will work, but if you are using a camera, then this would cause a conflict. A good alternative pin to use here would be PWM1.

A0 and A1 are meant to be only used as input pins and for ADC. The MicroMod Artemis Processor has op amps and scaling from 3.3V to 2V on both of those pins:

[image]

Hope this helps, and again, sorry for the delay in our response here.

Going to mark this issue as resolved. Please let us know if you run into any further problems.

Thanks,
Pete

closed 2024-07-29 by lewispg228

diplodocuslongus, 2024-07-30:
Hey Pete, thanks a lot!

Ludo