sparkfun/SparkFun_BNO08x_Arduino_Library #20: BNO08x::begin() hangs when trying to return _init()

Donato-M, 2024-06-06:
### Subject of the issue
Running the example 1 code does not work. Upon testing I have found the issue to be in the begin() method. It runs up until it reaches the final return point where it is supposed to return _init().

### Your workbench
* What development board or microcontroller are you using?
* ESP32-WROOM-32E
* What version of hardware or breakout board are you using?
* How is the breakout board to your microcontroller?
* The microcontroller is connected to my laptop via a micro-USB cable and the ESP32 Tester V1.1
* How is everything being powered?
*Power is supplied via the USB cable
* Are there any additional details that may help us help you?
* The BNO is soldered to a custom made PCB with all the necessary capacitors and resistors as specified in the BNO's datasheet for I2C communication.
* The BNO085 is connected to an I2C bus along with a number of other device. Communicating with these devices doesn't pose any problems

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.
I attempted to get the BNO working using example 1. I changed the pin numbers and the I2C address to match the connections on the PCB the BNO is on.

#define BNO08X_INT  27
#define BNO08X_RST  12
#define BNO08X_ADDR 0x4A // Alternate address if ADR jumper is closed

### Expected behavior
The Rotation vector example should execute

### Actual behavior
The program halts at the return _init() line of begin().

````
/*
  Using the BNO08x IMU

  This example shows how to output the i/j/k/real parts of the rotation vector.
  https://en.wikipedia.org/wiki/Quaternions_and_spatial_rotation

  By: Nathan Seidle
  SparkFun Electronics
  Date: December 21st, 2017
  SparkFun code, firmware, and software is released under the MIT License.
	Please see LICENSE.md for further details.

  Originally written by Nathan Seidle @ SparkFun Electronics, December 28th, 2017

  Adjusted by Pete Lewis @ SparkFun Electronics, June 2023 to incorporate the
  CEVA Sensor Hub Driver, found here:
  https://github.com/ceva-dsp/sh2

  Also, utilizing code from the Adafruit BNO08x Arduino Library by Bryan Siepert
  for Adafruit Industries. Found here:
  https://github.com/adafruit/Adafruit_BNO08x

  Also, utilizing I2C and SPI read/write functions and code from the Adafruit
  BusIO library found here:
  https://github.com/adafruit/Adafruit_BusIO

  Hardware Connections:
  IoT RedBoard --> BNO08x
  QWIIC --> QWIIC
  A4  --> INT
  A5  --> RST

  BNO08x "mode" jumpers set for I2C (default):
  PSO: OPEN
  PS1: OPEN

  Serial.print it out at 115200 baud to serial monitor.

  Feel like supporting our work? Buy a board from SparkFun!
  https://www.sparkfun.com/products/22857
*/

#include <Wire.h>

#include "SparkFun_BNO08x_Arduino_Library.h"  // CTRL+Click here to get the library: http://librarymanager/All#SparkFun_BNO08x
BNO08x myIMU;

// For the most reliable interaction with the SHTP bus, we need
// to use hardware reset control, and to monitor the H_INT pin.
// The H_INT pin will go low when its okay to talk on the SHTP bus.
// Note, these can be other GPIO if you like.
// Define as -1 to disable these features.
#define BNO08X_INT  27
//#define BNO08X_INT  -1
#define BNO08X_RST  12
//#define BNO08X_RST  -1

//#define BNO08X_ADDR 0x4B  // SparkFun BNO08x Breakout (Qwiic) defaults to 0x4B
#define BNO08X_ADDR 0x4A // Alternate address if ADR jumper is closed

void setup() {
  Serial.begin(115200);
  
  while(!Serial) delay(10); // Wait for Serial to become available.
  // Necessary for boards with native USB (like the SAMD51 Thing+).
  // For a final version of a project that does not need serial debug (or a USB cable plugged in),
  // Comment out this while loop, or it will prevent the remaining code from running.
  
  Serial.println();
  Serial.println("BNO08x Read Example");

  Wire.begin(21, 22, 350000);

  //if (myIMU.begin() == false) {  // Setup without INT/RST control (Not Recommended)
  if (myIMU.begin(BNO08X_ADDR, Wire, BNO08X_INT, BNO08X_RST) == false) {
    Serial.println("BNO08x not detected at default I2C address. Check your jumpers and the hookup guide. Freezing...");
    while (1)
      ;
  }
  Serial.println("BNO08x found!");

  // Wire.setClock(400000); //Increase I2C data rate to 400kHz

  setReports();

  Serial.println("Reading events");
  delay(100);
}

// Here is where you define the sensor outputs you want to receive
void setReports(void) {
  Serial.println("Setting desired reports");
  if (myIMU.enableRotationVector() == true) {
    Serial.println(F("Rotation vector enabled"));
    Serial.println(F("Output in form i, j, k, real, accuracy"));
  } else {
    Serial.println("Could not enable rotation vector");
  }
  delay(100);
}

void loop() {
  delay(10);

  if (myIMU.wasReset()) {
    Serial.print("sensor was reset ");
    setReports();
  }

  // Has a new event come in on the Sensor Hub Bus?
  if (myIMU.getSensorEvent() == true) {

    // is it the correct sensor data we want?
    if (myIMU.getSensorEventID() == SENSOR_REPORTID_ROTATION_VECTOR) {

      float quatI = myIMU.getQuatI();
      float quatJ = myIMU.getQuatJ();
      float quatK = myIMU.getQuatK();
      float quatReal = myIMU.getQuatReal();
      float quatRadianAccuracy = myIMU.getQuatRadianAccuracy();

      Serial.print(quatI, 2);
      Serial.print(F(","));
      Serial.print(quatJ, 2);
      Serial.print(F(","));
      Serial.print(quatK, 2);
      Serial.print(F(","));
      Serial.print(quatReal, 2);
      Serial.print(F(","));
      Serial.print(quatRadianAccuracy, 2);

      Serial.println();
    }
  }
}
```
![image](https://github.com/sparkfun/SparkFun_BNO08x_Arduino_Library/assets/110125544/36b6dae8-abe1-4bd7-890b-c66bbf7cf0b3)

Donato-M, 2024-06-06:
Upon poking around in the library files I've found the issue to be in the sh2_getProdIds method, located in the sh2.c file. I'll be honest and say I have no idea what this method is doing and can't even figure out where to begin trying to understand why I'm encountering an error

closed 2024-06-11 by Donato-M

MickTheMechanic, 2025-06-06:
Im experiencing the same issue, did you figure out what was happening?